RoganDawes/P4wnP1_aloa #328: Not a single thing that matter works anymore, don't waste your time and energy on this

scrypthekid, 2023-04-10:
Except the web UI working and that it's possible to change startup configurations, nothing that actually matters such as USB tethering, shared connection of any kind, launching HID scripts or anything else that made the project interesting or worthwhile works on any system: I've spent an ungodly amount of time trying to get it to work on Windows 7-11, a bunch of Debian based distros, Manjaro and Arch and I couldn't get anything to work anywhere. It won't even get recognized as mass storage or CD-Rom.
The only things working are:
-The web UI out of the box (WIFI only tho) and only some configurations
-SSH only via the WIFI server

I didn't even bother trying out anything bluetooth related since the most basic functions where a frustrating nightmare.
It's a damn shame this project got deserted like this...

scoull, 2023-04-11:
Download the Kali version.

It's only for Pi Zero W but I can't find a single thing that doesn't work after spending weeks of trying to get USB gadgets to work with Raspberry's own OS.

https://www.kali.org/docs/arm/raspberry-pi-zero-w-p4wnp1-aloa/

scrypthekid, 2023-04-11:
Appreciate the input! I've tried flashing the repo image, the original kali image and even the one included in Rpi-Imager, but like I said it just doesn't work; none of the HID scripts execute on any of my systems, whether on bare metal or VM.
I found [this](https://github.com/woahotter/pwnhyve) interesting project yesterday at 3 a.m. which I'll probably give a try soon, it also includes an OLED hat with embedded controls for switching modes/scripts and seems pretty interesting. Also if you made yours work with the RoganDawes repo, you should check [this project](https://gideonwolfe.com/posts/security/p4wnp1/) out if you want to add a screen with controlls. Cheers!