RoganDawes/P4wnP1_aloa #331: HOW TO COMPILE IT FOR RASPBERRY PI ZERO 2W

circuito-suman, 2023-05-15:
ERROR /root/go/pkg/mod/golang.org/x/net@v0.10.0/http2/transport.go:19:2: package io/fs is not in GOROOT (/usr/lib/go-1.15/src/io/fs)
service/pgpio/p4wnp1gpio.go:10:2: module periph.io/x/periph@latest found (v3.7.0+incompatible), but does not contain package periph.io/x/periph/conn/gpio
service/SubSysGpio.go:11:2: module periph.io/x/periph@latest found (v3.7.0+incompatible), but does not contain package periph.io/x/periph/conn/gpio/gpioreg
service/pgpio/p4wnp1gpio.go:11:2: module periph.io/x/periph@latest found (v3.7.0+incompatible), but does not contain package periph.io/x/periph/conn/physic
service/SubSysGpio.go:12:2: module periph.io/x/periph@latest found (v3.7.0+incompatible), but does not contain package periph.io/x/periph/conn/pin/pinreg
service/SubSysGpio.go:13:2: module periph.io/x/periph@latest found (v3.7.0+incompatible), but does not contain package periph.io/x/periph/host
service/SubSysGpio.go:14:2: module periph.io/x/periph@latest found (v3.7.0+incompatible), but does not contain package periph.io/x/periph/host/rpi

lgeekjopt, 2023-06-04:
You can use my build, compiled for pizero 2w
https://github.com/lgeekjopt/P4wnP1_aloa/releases/tag/pi_zero2w

maroc46, 2023-12-31:
> You can use my build, compiled for pizero 2w
> 
> https://github.com/lgeekjopt/P4wnP1_aloa/releases/tag/pi_zero2w

Hey lgeekjopt/Joaquim I Hope you can help me .
You have the p4wnp1 Aloa rewritten that is compitable with the piZero2W 

My question is because I want to use a piZero2W with kali to connect it with usb-Ethernet on my windows pc via ssh 

But in my case 
I don't want to use it with wiFi 
Only with otg usbeth 

What I must doing for 
I have seen that I must change a file in config.txt 

But I need a rndis driver to connect it with my pc ? 

I find nothing 

I read that linux deleted from linux Kernel 
In Kali-tweaks if find nothing 

IHope you can help me