RoganDawes/P4wnP1_aloa #332: This is not an issue, but why is the project dead?

Crash891, 2023-07-10:
This project works so good for me, using the pi zero 2 version (down in the issues), and permit me to do lots of stuff... i don't manage to 
understand why this project is dead while it's working so good

clout86, 2023-07-10:
There appears to be people asking this from time to time. I think there
might be an upstream dev source somewhere though not sure the state of it
(it's been awhile), but I would probably call it stale rather than dead.

Maybe fork your working version or see if you can push changes?

Magnusretrotech, 2024-01-30:
> This project works so good for me, using the pi zero 2 version (down in the issues), and permit me to do lots of stuff... i don't manage to understand why this project is dead while it's working so good

Can you please fork the version you got working on the pi zero 2 w and send it to me? I have a 2 and really want to use it.

quesodipesto, 2024-04-10:
> This project works so good for me, using the pi zero 2 version (down in the issues), and permit me to do lots of stuff... i don't manage to understand why this project is dead while it's working so good

Hi Mate! Can u share with us/me your version? Thanks bro!! :D

RoganDawes, 2024-11-05:
It's "dead" because the original dev (MaMe82) prioritised his family over hacking on this and other projects. He handed his projects over to me (RoganDawes) as a caretaker, but I also don't have time and, more importantly, knowledge to maintain them. My Go skills are miniscule, and some of Marcus's choices make the project more difficult to approach (particularly the GUI/web library).

closed 2024-11-05 by RoganDawes