RoganDawes/P4wnP1_aloa #339: OpenStick question

wvthoog, 2023-11-29:
Hi, awesome project which i'm trying to get running on UF896 OpenStick

I've created a Debian Bookworm chroot environment and installed Go 1.17.9. Compiled it like so

```
apt install wget hostapd iproute2 iw
wget https://go.dev/dl/go1.17.9.linux-arm64.tar.gz
tar -C /usr/local -xzf go1.17.9.linux-arm64.tar.gz
export PATH=$PATH:/usr/local/go/bin
go install github.com/gopherjs/gopherjs@v1.17.2

git clone https://github.com/RoganDawes/P4wnP1_aloa.git
cd P4wnP1_aloa/build_support
chmod +x build.sh && ./build.sh
```

That build succesfully and copied the files over like so

```
cp build/P4wnP1_service /usr/local/bin/
cp build/P4wnP1_cli /usr/local/bin/
cp dist/P4wnP1.service /etc/systemd/system/P4wnP1.service
mkdir -p /usr/local/P4wnP1
cp -R dist/keymaps /usr/local/P4wnP1/
cp -R dist/scripts /usr/local/P4wnP1/
cp -R dist/HIDScripts /usr/local/P4wnP1/
cp -R dist/www /usr/local/P4wnP1/
cp -R dist/db /usr/local/P4wnP1/
cp build/webapp.js /usr/local/P4wnP1/www
cp build/webapp.js.map /usr/local/P4wnP1/www
```

Booted the OpenStick and started the P4wnP1 service

```
● P4wnP1.service - P4wnP1 Startup Service
     Loaded: loaded (/etc/systemd/system/P4wnP1.service; disabled; preset: enabled)
     Active: active (running) since Wed 2023-11-29 18:12:38 CET; 2s ago
   Main PID: 2395 (P4wnP1_service)
      Tasks: 12 (limit: 411)
     Memory: 108.7M
        CPU: 1.171s
     CGroup: /system.slice/P4wnP1.service
             ├─2395 /usr/local/bin/P4wnP1_service
             ├─2449 /usr/sbin/hostapd /tmp/hostapd_wlan0.conf
             └─2463 /usr/sbin/dnsmasq --log-facility=- -k -x /var/run/dnsmasq_wlan0.pid -C /tmp/dnsmasq_wlan0.conf

nov 29 18:12:40 openstick P4wnP1_service[2395]: 2023/11/29 18:12:40 Stopping DHCP client for interface 'wlan0'...
nov 29 18:12:40 openstick P4wnP1_service[2395]: 2023/11/29 18:12:40 Setting Interface wlan0 to IP 172.24.0.1
nov 29 18:12:40 openstick P4wnP1_service[2395]: 2023/11/29 18:12:40 Setting Interface wlan0 to UP
nov 29 18:12:40 openstick P4wnP1_service[2395]: 2023/11/29 18:12:40 Setting Interface wlan0 to MULTICAST to ON
nov 29 18:12:40 openstick P4wnP1_service[2395]: 2023/11/29 18:12:40 ... killing dnsmasq
nov 29 18:12:40 openstick P4wnP1_service[2395]: 2023/11/29 18:12:40 ... dnsmasq for interface 'wlan0' isn't running, no need to stop it
nov 29 18:12:40 openstick P4wnP1_service[2395]: 2023/11/29 18:12:40 Starting dnsmasq for interface 'wlan0' with config '/tmp/dnsmasq_wlan0.conf'...
nov 29 18:12:40 openstick P4wnP1_service[2395]: 2023/11/29 18:12:40 ... DHCP server for interface 'wlan0' started
nov 29 18:12:40 openstick P4wnP1_service[2395]: 2023/11/29 18:12:40 SetLEDSettings blink_count:2
nov 29 18:12:40 openstick P4wnP1_service[2395]: /usr/local/P4wnP1/scripts/servicestart.sh: LED blink count set to 2
```

I can access the Wifi access point, get an IP address assigned and go to the webpage at http://172.24.0.1:8000

But then the only thing i see is a pulsating circle. So opened Chrome console which pointed me to an error in webapp.js. The exact error message:

```
Uncaught TypeError: Cannot read properties of undefined webapp.js:28285 (reading 'length')
```

Specifcally at this section

```
this.arrClean=!0,0<this.arr_.length&&this.loadFromArray_()}}
```

Maybe the array is empty ?

Any pointers on how to solve this are more than welcome. 

Thanks !

wvthoog, 2023-12-01:
What i've also tried is just build webapp.js

```
git clone https://github.com/RoganDawes/P4wnP1_aloa.git
cd P4wnP1_aloa/web_client
gopherjs build -o ../www/webapp.js
```

and copied those files _(webapp.js and webapp.js.map)_ over to _/usr/local/P4wnP1/www/_

And also tried just using the shipped webapp.js and webapp.js.map from the Git build/ directory.

All with the same result: Uncaught TypeError

```
Uncaught TypeError: Cannot read properties of undefined webapp.js:28185 (reading 'length')
```

What did work though is downloading the v0.1.1-beta release and extracting webapp.js and webapp.js.map and copying those files over. The webinterface loads correctly

gjfmlj, 2024-08-12:
How to install it on OpenStick, and has anyone successfully installed it?