RoganDawes/P4wnP1_aloa #340: P4wnP1_aloa not running on Raspberry Pi Zero W

markusruebe, 2023-12-01:
Hi, I try to follow the c't article to test "bad USB" with a Pi pico and an USB board.
I downloaded the P4wnP1_aloa zip from GitHub and burned it to a micro SD card with piImager.
There should be a WiFi AP published.
There is a blue LED of the USB board but the green LED for micro SD does not flash. I used different cards that all work on raspi 4 / 400 but nothing changes.
[image]
[image]
Any idea? Thanks for some hints.

Hjack737, 2023-12-26:
Did you connect it over the usb thing there, ore did you try to connect it via micro usb and the usb port?

MartinMainusch, 2024-01-24:
I also followed the c't article and at first it did not work on my device either. But after I downladed the correct release [v0.1.1-beta](https://github.com/RoganDawes/P4wnP1_aloa/releases/tag/v0.1.1-beta) it worked. Previously I made the blunder to download the source code instead of the release image.

linguine2552, 2024-02-29:
can you plug it into a monitor to see what the boot sequence is halting at? I was able to get this project working back in 2021 and the release is still the same but now my pi is having bluetooth interface errors. If this is the case for others I can fork and build a new image.

mrfoxie, 2024-03-09:
will it work on raspberry pi 0 2w

tikoker, 2025-02-13:
I tried the latest release(v0.1.1-beta) on different sd cards. It does not work on raspberry pi 0 2w. I tried downloading "balera etcher" and "rufus". It only gives out this picture:

[image]